riscv-non-isa/riscv-server-platform #2: Mandate IOMMU RIMT (ACPI) requirement

andreiw, 2024-06-03:
AFAICT Server SoC mandates IOMMU
BRS has non-normative text about RIMT on systems supporting IOMMU

...need requirement in *this* spec to tie it together.

andreiw, 2024-07-29:
Actually such a text could well be in the BRS document, but isn't (along with many other requirements that are implicit in the ACPI spec and dependent specs).

So do we really need the RIMT text (i.e. if you have an IOMMU, you will expose it via RIMT). Why would IOMMU get this treatment and something else (AIA) would not?

andreiw, 2024-07-29:
The thought here is that ACPI _doesn't_ actually require a RIMT for a RISC-V compliant IOMMU.

andreiw, 2024-07-29:
The BRS does not really have normative text around the RIMT. A solution could be to make RIMT spec one of the BRS-I dependencies (and ensure RIMT has text that requires implementation if a compliant RISC-V IOMMU is present)

jones-drew, 2025-01-20:
While the BRS does have requirements dictating which tables to use, like ACPI_030 which states a PPTT must be used even for a system with a simple hart topology, conditional requirements like "If there's an IOMMU, then the RIMT must be used" would require the BRS to consider what there might be. The platform server spec is better positioned for that since it's sure there may be an IOMMU, as it also builds on the server-soc. So, if there's any concern that a RIMT won't be used when there's an IOMMU, then I think we should have a firmware requirement for it.

andreiw, 2025-01-23:
> While the BRS does have requirements dictating which tables to use, like ACPI_030 which states a PPTT must be used even for a system with a simple hart topology, conditional requirements like "If [...]

I think it's fine to land it here, for now anyway. The reason it isn't in the BRS is because it wasn't done, and we didn't want another dependency on something that wasn't done (but clearly we were okay lounging around for UEFI and ACPI updates, heh). There is a precedent for "if this then that" rules - we have one for the TPM there.

BRS did have a ticket (https://github.com/riscv-non-isa/riscv-brs/issues/18) but I closed it so we could converge the spec on something ratifiable (and open tickets made it seem like we would never be done). I'll reopen it for visibility. If this gets pulled into BRS, we can clean it up here.

closed 2025-02-03 by jones-drew