riscv-non-isa/riscv-server-platform #4: Add SBI requirements for server platforms that are a superset of BRS ones

andreiw, 2024-06-03:
Looking at SBI 3.0 extensions, wherever required and matching Server SoC hardware requirements (e.g. RAS)

andreiw, 2024-06-03:
Another example is hosted/native debug extension.

Need to scan the Server SoC requirements and identify the new SBI requirements that don't exist in the BRS.

andreiw, 2024-06-17:
For RAS looks like we just need https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/src/ext-sse.adoc so far....

jones-drew, 2025-01-20:
> Another example is hosted/native debug extension.

Should we require support for hosted/native debug?

> 
> Need to scan the Server SoC requirements and identify the new SBI requirements that don't exist in the BRS.

We may need FWFT, depending on what security features we require.

For now, I've created a PR with only SSE called out. We should discuss DBTR and FWFT more.

jones-drew, 2025-02-03:
Merging the PR automatically closed this issue, but we may need to add more.

jones-drew, 2025-06-16:
We shouldn't need to call out FWFT since nothing it can currently control is mandated. We should mandate DBTR, though, since we already mandate sdtrig.

jones-drew, 2025-06-26:
Resolved by PR #72

closed 2025-06-26 by jones-drew