riscv-non-isa/riscv-server-platform #11: firmware: MUST boot in HS mode if virtualization is implemented

andreiw, 2024-06-14:
This allows a VM implementation but on a host with virtualization support you shouldn't have surprises.

andreiw, 2024-06-14:
BRS has this text in the non-normative portion: "UEFI implementations run in 64-bit S-Mode, VS-Mode or HS-Mode, depending on whether virtualization is supported or used."

Again, it's probably a good idea to make this a stronger req.

jones-drew, 2025-01-20:
I agree we should strengthen this. Since we've decided not to consider VM targets for this spec, then we can avoid the complications of deciding whether or not we require nested virt support and just focus on S-mode vs. HS-mode. The server spec should mandate HS-mode.

jones-drew, 2025-04-07:
Closing this issue with the justification written in https://github.com/riscv-non-isa/riscv-server-platform/pull/27#issuecomment-2783971399

closed 2025-04-07 by jones-drew