riscv-non-isa/riscv-server-platform #12: Firmware: require iSCSI boot support on machines lacking persistent storage.

andreiw, 2024-06-14:


xypron, 2024-06-17:
Looking at the move in industry to NVMEoF, wouldn't this be a valid alternative?

andreiw, 2024-07-29:
We could certainly phrase it in an agnostic manner.

andreiw, 2024-07-29:
After working through UEFI section 2.6.2 (platform specific elements) I think it's silly to /require/ network boot from network block devices. In the end, devices built for a specific part of the server market will include such functionality (and where they don't, third party alternatives exist like iPXE)..

So far https://github.com/riscv-non-isa/riscv-server-platform/pull/20 added disk and network boot as a SHOULD. I would anticipate most implementations to take this in earnest, and possibly some cloud implementations could omit/simplify to meet their infra requirements.

closed 2024-07-29 by andreiw

xypron, 2024-07-29:
Am 29. Juli 2024 03:27:55 MESZ schrieb Andrei Warkentin ***@***.***>:
> After working through UEFI section 2.6.2 (platform specific elements) I think it's silly to /require/ network boot from network block devices. In the end, devices built for a specific part of the [...]

iPXE does not suport RISC-V, yet.

Software like iPXE will at least need the simple network protocol.

If the server firmware is based on EDK II, you will be able to fulfill the requirements of the 2.6.2 section if you have a network driver. The only platform specific work is adding the NIC driver. 

Best regards

Heinrich

>
>So far https://github.com/riscv-non-isa/riscv-server-platform/pull/20 added disk and network boot as a SHOULD. I would anticipate most implementations to take this in earnest, and possibly some cloud implementations could omit/simplify to meet their infra requirements.
>