riscv-non-isa/riscv-server-platform #18: Questions about security requiremements

andreabolognani, 2024-06-27:
First, a simple clarification: the spec says

>  it is expected the high-level RoT / boot flow requirements

What is "RoT" in this context? I'm not familiar with the acronym and it's not defined in the glossary, nor was I able to quickly find an explanation online. It doesn't help that "rot" is a common word in the English language :)

More to the point, I see that `SEC_010` mandates that UEFI Secure Boot is implemented. Is that actually feasible today?

I was recently discussing the RISC-V implementation of shim, which is a critical part of how Linux implements the Secure Boot trust chain, and a concern [was raised](https://github.com/rhboot/shim/pull/641#issuecomment-2185830943) that the necessary infrastructure/agreements might not be in place yet. Do we have additional information about that?

Or is the intention that the *mechanisms* for Secure Boot are implemented, with no expectation that Microsoft's (or anyone else's) key is actually enrolled out of the box? If so, would it be worth clarifying in the non-normative text?

andreiw, 2024-07-09:
RoT in this context is "root of trust".

UEFI Secure Boot should be feasible, why not? Do you have indications to the contrary? For example, there are multiple explored mechanisms for implementing authenticated variables (the largest dependency, as I see it... some of this work has been done within RISE FW group). Of course, the proof is the pudding, which is why we have the SiP presence in the group.

Wrt https://github.com/rhboot/shim/pull/641#issuecomment-2185830943, why wouldn't MS sign it? There haven't been any indications with prior efforts that this could become a problem. If it becomes a problem, then it simply accelerates moving to a separate authority. This is something I can follow up with or clarification or at least a statement.

andreabolognani, 2024-07-15:
> RoT in this context is "root of trust".

Gotcha. I would expand the acronym for clarity but since it's in a paragraph marked "TBD" that might be seen as unnecessary churn, so I'm holding off for now.

> UEFI Secure Boot should be feasible, why not? Do you have indications to the contrary? For example, there are multiple explored mechanisms for implementing authenticated variables (the largest [...]

With SiP meaning... ? O:-)

I have no indication that Secure Boot can't be achieved, quite the contrary: I would consider lack of Secure Boot for a server platform a near fatal flaw. I'm just wondering whether one could ship an implementation *today*. The fact that you mention "multiple explored mechanisms" makes me think that it's not a solved problem. And so I wonder what that means for the Server Platform spec... Do we want to mandate something that we aren't 100% sure can actually be implemented?

> Wrt [rhboot/shim#641 (comment)](https://github.com/rhboot/shim/pull/641#issuecomment-2185830943), why wouldn't MS sign it? There haven't been any indications with prior efforts that this could become a problem. If it becomes a problem, then it simply accelerates moving to a separate authority. This is something I can follow up with or clarification or at least a statement.

Same as above, I have no indication that Microsoft would refuse to sign riscv64 shims when they're already signing x86_64/aarch64 shims. But I have no indication that they definitely *are* going to sign them either. Do you?

I'm not overly familiar with the (massive) UEFI spec, so I'm not sure whether it actually contains any requirement to ship the Microsoft keys out of the box. Maybe it doesn't, and Microsoft being the de-facto gatekeeper for the Secure Boot ecosystem is more of a reflection of reality than something that's mandated by spec. If so, good! It means that we can require vendors to implement Secure Boot regardless of whether Microsoft is up for signing additional shims.

There's another thought that popped into my mind after I had filed this issue. There was a controversy [a decade ago](https://www.theverge.com/microsoft/2012/1/16/2710502/microsoft-secure-boot-windows-8-arm) about Microsoft's certification program requiring that Arm devices are locked down in terms of Secure Boot, specifically preventing it from being turned off or (IIRC) enrolling custom keys.

It looks like Windows 10 further [moved in the wrong direction](https://www.howtogeek.com/116569/htg-explains-how-windows-8s-secure-boot-feature-works-what-it-means-for-linux/#what-microsoft-requires-of-pc-manufacturers) by making it possible, even on x86_64, for vendors to ship hardware that doesn't provide a way for the user to turn Secure Boot off and doesn't include the Microsoft third-party key (the one used to sign shim).

I don't know if things have changed again in the meantime, but I wonder: would it make sense to make it an explicit requirement in the Server Platform spec that the user must be able to turn Secure Boot off and enroll their own keys? I feel like both are very desirable properties.

I would actually argue that such a requirement might belong in the BRS spec, but we could start by making it part of the Server Platform spec and try pushing it down the stack later.

andreiw, 2024-07-23:
SiP -> Silicon Providers.

Authenticated variables definitely /can/ be implemented, but the actual underlying mechanism of isolating the MM kernel is SoC and possibly even vendor micro-architecture specific. Here are the three mechanisms that I am aware of:
- Running MM on a System Control Processor, microcontroller or non-"rich OS" capable hart, with UEFI RT and MM communicating via RPMI. The flash chip in this case belongs to the SCP/MCU running the MM kernel and is not accessible to normal "rich OS" harts.
- Running MM in a vendor-specific "secure enclave" on normal "rich OS" capable harts. The "security visor" protects the flash from the regular HS/S modes.
- Running MM in M mode.

Regardless of how it is implemented, there would be no OS-observable differences in behavior.

As far as consumer Arm devices, yes some unfortunate history in devices that were an evolutionary dead end (Windows RT), but that hasn't leaked into server systems endorsed by MS nor even more recent devices. I agree with the sentiment of requiring the ability to disable Secure Boot and to enroll own keys. Those are basically required for feature parity with Arm and x86.

andreiw, 2024-07-29:
I reached out to uefisign@microsoft.com

andreabolognani, 2024-07-29:
@andreiw thanks for reaching out to Microsoft.

In the meantime, I've looked at the Windows 11 "Hardware Compatibility Specification" and taken inspiration for our own requirements in terms of keeping Secure Boot fully manageable by users. We could go a lot more in depth if desired, but for now I've decided to keep things very simple.

andreiw, 2024-07-29:
Merged https://github.com/riscv-non-isa/riscv-server-platform/pull/22

andreiw, 2024-11-10:
Okay I've heard back from uefisign, apparently there was some reorgnanization and the earlier email fell through the seams.

andreiw, 2024-11-10:
Now waiting for their input...

jones-drew, 2025-01-20:
> Now waiting for their input...

Hi Andrei, any word from MS?
Thanks, drew

rsahita, 2025-03-17:
Hi @jones-drew - I have the contact names from @rpsene for MS now - will send a note and keep you posted.

rsahita, 2025-05-02:
update - Microsoft has responded and will let us know after they check internally.

jones-drew, 2025-06-16:
> update - Microsoft has responded and will let us know after they check internally.

Hi @rsahita, any update on this? Thanks, drew

andreiw, 2026-10-01:
Let me try to pick this up where I left it off... which is to reach out to uefisign@microsoft.com

andreiw, 2026-10-02:
I reached out. Waiting.

andreiw, 2026-10-05:
Had this discussion at server platform meeting on Oct 5, and Robin Maffeo (MS) volunteered to look further internally. The main question was whether there *was* something to sign, and Andrea Bolognani explained that while no package exists yet, the repo can be manually built.

rpsene, 2026-10-05:
> Had this discussion at server platform meeting on Oct 5, and Robin Maffeo (MS) volunteered to look further internally. The main question was whether there _was_ something to sign, and Andrea Bolognani explained that while no package exists yet, the repo can be manually built.

@andreiw  Not sure if this happened on not, but now we would need @andrewdellow input here.

andreiw, 2026-10-06:
> [@andreiw](https://github.com/andreiw) Not sure if this happened on not, but now we would need [@andrewdellow](https://github.com/andrewdellow) input here.

Acknowledged. I will reach out.