riscv-non-isa/riscv-server-platform #24: Server Platform Ecosystem Requirements

andreiw, 2024-07-29:
- Require upstream Linux support (some platforms may meet the letter of the spec, but still be in practice unusable without some changes that are not in upstream, but present in some distro or private fork)
- (maybe) require (some proof of) downstream engagement?

jones-drew, 2025-01-20:
Hi Andrei,

Since upstream is a moving target it would be tough to write a testable requirement for this. Also, even though it's certainly preferred that upstream Linux will always work on server platforms, as long as popular OS vendors are supporting them, then I think this specification is serving its purpose.

Do you agree we should close this issue? Or do you have some other thoughts on it?

Thanks, drew

jones-drew, 2025-04-08:
I'll close this one since I don't think we can write a testable requirement for it.

closed 2025-04-08 by jones-drew