riscv-non-isa/riscv-server-platform #32: Mandate hardware support for misaligned access handling

andreiw, 2025-01-28:
There's nothing in an ISA profile that mandates behavior (see https://github.com/riscv/riscv-profiles/issues/187).

While the server platform requirements don't specifically have /performance/ targets (just like they don't about electricals, mechanicals, etc):
- The expected software stack ("server software stack" or "general purpose compute") comes from an environment where unaligned accesses penalty isn't in the range of 2-3 factors.
- the spec already mandates performance-oriented extensions like Svadu, which translates to a system with less "maintenance traps" taken to OS kernel.

Mandating hw support for unaligned accesses would obviously go a long stretch to reducing more of these "maintenance traps" to M or HS/VS modes (if delegated). We don't have to talk about the performance of the hw implementation (just like we don't elsewhere)

You could argue that this is unenforceable, but that depends what you mean:
a) an OS or format test suite can easily detect unaligned trap behavior (i.e. use FWFT to prevent M mode from handling, or just measure the performance like the code linked off of https://github.com/riscv/riscv-profiles/issues/187 does... and IIRC Linux already attempts to detect this!)
b) formal certification, if that ever comes to pass, could be refused on the basis of any tests failing

jones-drew, 2025-04-08:
If we're not going to talk about performance, then I think the requirement to support Zicclsm is the best we can do. Unfortunately it's not 100% clear that Zicclsm means there won't be traps to M-mode, since the profile behavior is defined for U/S-mode. However, platforms targeting server workloads will likely see the need to support Zicclsm with hardware and even try to make it fast. And, in any case, if we want to mandate "hw misaligned access handling", then I guess we should try to get an extension / named feature created for it and then mandate the inclusion of that extension by its name in this spec. I'll close this issue for now with my fingers crossed that Zicclsm is sufficient.

closed 2025-04-08 by jones-drew

sorear, 2025-09-17:
The new named feature seems to be [Oilsm](https://riscv.atlassian.net/wiki/external/ZGZjMzI2YzM4YjQ0NDc3MmI3NTE0NjIxYjg0ZGJhY2E).