riscv-non-isa/riscv-server-platform #51: v0.6: RVA_021 Justify Ssctr

jones-drew, 2025-05-05:
Terje Bergstrom has asked for justification for Ssctr. Beeman Strong writes:

"""
Similar capabilities from other architectures (x86's LBRs,
ARM's BRBE) are used by profile-guided optimization (PGO) tools like
autofdo, BOLT, and propeller to enrich branch profiles.  These tools
typically observe gains in the 5-20% range by improving builds over time.
Hyperscalers like Google and Meta use these tools heavily and are driving
their development.
"""

jones-drew, 2025-05-21:
@bcstrongx please see the inquiry above.

bcstrongx, 2025-05-21:
While one may be able to use E-trace (or N-trace) to produce branch profiles for PGO tools, I don't think we would want to make this an option for the server platform spec, for the following reasons.

1) There is no standard ISA interface to trace for RISC-V, all ratified trace standards are non-ISA.  They are more akin to ARM's ETM, tailored for external debug use.  There is a Self-hosted Trace TG in progress that will add an ISA interface in time.
2) Trace to memory adds a few percent performance overhead during profile generation, and the requirement to decode trace adds an extra step (and latency) to the process.
3) I don't know of any work to support E-trace or N-trace in Linux perf currently.

There was some work to use ARM ETM to feed autofdo profiles, particularly for Android and ChromeOS.  My understanding is that adoption was limited for the reasons listed above, and because ETM is frequently unavailable in production parts.  There were evidently also security concerns, but I'm less familiar with them.  The fact that ARMv9-A added BRBE (analogous to CTR) I think is telling here.  ETM was a sub-optimal solution.

So rather than burdening software with supporting a second method for collecting the type of rich branch profiles needed for PGO, I'd stick to CTR.

closed 2025-06-02 by jones-drew