riscv-non-isa/riscv-server-platform #83: What about WARL fields and other choices?

tariqkurd-repo, 2025-07-29:
Referring to this statement, and the following paragraph:

https://github.com/riscv-non-isa/riscv-server-platform/blob/2e7bbad158e67f6a446718ed8c03f1a59e927b15/server_platform_requirements.adoc?plain=1#L44

Is it really enough that the same extensions are implemented?
Doesn't there need to be much greater alignment?
For example WARL field handling varies greatly between implementations.
Does this statement assume that all WARL fields are the same in the hardware implementations, or that the software running correctly understands all possible WARL field settings and so can run identically on any valid implementation?

Is this level of detail given in any of the specifications?

radimkrcmar, 2025-07-30:
A wording like "the set of legal values in CSRs must be the same on all harts" is possibly better.
I think that `RVA_050` covers it, though, because it has very strong requirement of indistinguishable harts:
```
The RISC-V application processor harts in the SoC MAY support
different power and performance characteristics but MUST be
otherwise indistinguishable from each other from a software
execution viewpoint within a supervisor execution environment (SEE).
```

radimkrcmar, 2025-07-30:
I don't think it's better specified anywhere, but the exact meaning should be that there must not exist a program that can prove that its different behavior is caused by the hart it's executing on, independent of hart's performance characteristic.

I would hope that when something is as informal as the quoted specification, implementations should be as strict as possible, and software should try to be lenient, to always meet somewhere in the middle.

For `RVA_040`/`RVA_050`, the harts theoretically don't even have to be the same mvendorid+marchid+mipid -- a server platform can combine different chips if M-mode can eliminate any behavioral difference between them (e.g. by disabling features to match the intersection, and defining the resulting configuration appropriately).
A lot of RISC-V behavior cannot be architecturally controlled, and quirks of any chip would have to be implemented on other chips as well, but a single vendor should be able to provide something like powerful/efficient cores of different marchid.

On the WARL example: an invalid write can result in any legal value being put into the CSRs.  The ISA specifies that the legal value must deterministically depend on the architectural state.
Software must not be able to prove that the legal value after an illegal write depends on `mhartid` or any other hart-dependent architectural state that S-mode software cannot otherwise observe.
(I think it's hard to prove, because time and performance counters are part of observable architectural state as well.)

In reality, the details of illegal WARL behavior will be irrelevant, because no sane software should ever write an illegal value (we might get there in a few years), and current software using WARL for discovery should check that after a write, the value is what it wrote, and write a known legal value if not, instead of blindly depending on the implementation behavior.

tariqkurd-repo, 2025-08-01:
`associated CSR field widths` from `RV_040` which covers ASID and VMID - you're right.

But yes - the physical address bits are not a CSR field and directly affect the OS for the page table mappings.
Can software really cope with mismatched physical address widths?

On that basis you need PMAs to match too (PMP are covered by `associated CSR field widths` - although the _number_ of regions isn't.... plus the WARL behaviour in the pmpcfg registers?).
But these are machine-mode and your statement is about supervisor-mode - but even so - these things must match mustn't they?

I.e. to run the same software identically on multiple cores, the PMAs and PMPs must at leave cover the required working set of the application with the same attributes.

radimkrcmar, 2025-08-01:
> Can software really cope with mismatched physical address widths?

Software shouldn't even notice unless the platform makes a mistake, and puts memory outside the physical address width of some harts.
(The main result of accessing memory that doesn't exist would be an exception, and I think the physical address width difference can only show in satp.mode=bare, which isn't really expected to be used by software.)

> But these are machine-mode and your statement is about supervisor-mode - but even so - these things must match mustn't they?
>
> I.e. to run the same software identically on multiple cores, the PMAs and PMPs must at leave cover the required working set of the application with the same attributes.

Yes.  That is covered by `RVA_050`'s `indistinguishable`.  The implementation (ISA + M-mode) just must make the SEE look the same on all harts.  How it's achieved is unspecified.  (The behavior of M-mode only CSRs/extensions is irrelevant for us.)

tariqkurd-repo, 2025-08-01:
ok - and clearly there must be enough PA bits for all cores to cover the physical address space for the OS to be able to allocate the memory - I'm mentioning this explicitly as it's not covered by any CSRs

jones-drew, 2025-08-26:
Resolved with PR #85

tariqkurd-repo, 2025-09-30:
I'm in favour of merging the rules.