riscv-non-isa/riscv-server-platform #89: DMA covering whole address space

xypron, 2025-08-26:
On some platforms DMA only covers low 4 GiB of physical memory. This requires drivers to copy each block of memory multiple times.

Should we requre that DMA covers the whole memory range?

radimkrcmar, 2025-08-26:
Server platform `HSOC_010` says:
```
RISC-V SoCs MUST comply with the Server SoC v1.0 specification.
```
The Server SoC specification has two related rules:
*  `IOM_020`:
```
All DMA capable peripherals (RCiEP and non-PCIe devices) and all PCIe root ports accessible
by software on the RISC-V application processor harts MUST be governed by an IOMMU.
```
* `IOM_050`
```
The IOMMU MUST implement all the page based virtual memory system modes and
extensions that are implemented by the RISC-V application processor harts in the SoC.
```
Is that insufficient?

jones-drew, 2025-09-05:
@xypron

closed 2025-12-01 by jones-drew