riscv-non-isa/riscv-server-platform #90: Debug requirements

pdonahue-ventana, 2025-08-27:
Are there reasons that most of the debug requirements from https://github.com/riscvarchive/riscv-platform-specs/blob/main/riscv-platform-spec.adoc#debug were dropped?  For instance, why is it not required that native/self-hosted debug be supported at all?

jones-drew, 2025-08-27:
Hi Paul,

Is requiring Sdtrig, Sdext, and the requirements listed in rules RVA_060 and RVA_070 insufficient? If so, can you please point to the ratified versions of what was dropped from the old list that you suggest adding [back]?

Thanks,
drew

ved-rivos, 2025-08-27:
In general we have refrained from mandating anything that portable operating systems or hypervisors can observe at privilege level less than M. None of Sdtrig or Sdext are visible to S-level ISA. Should we drop RVA_060/RVA_070 from the rules for this version and revisit this later when there is ISA support for S-level self hosted debug.

pdonahue-ventana, 2025-08-27:
In that case, RVA_060 and the mandate for Sext in RVA_020 seem inappropriate.  And you just updated your comment to say what I was going to say about Sdtrig visibility through SBI.  Much of what I said would apply if only action=0 is supported.

Would the Sdext requirement fall into a similar bucket as HPER_010?  Remote access and early boot software aren't visible to the OS/hypervisor, either.  There are lots of other things in this spec that aren't really visible to the OS/hypervisor, though one can argue that something like secure boot is visible to the OS because it ensures that you're actually running the right OS.

ved-rivos, 2025-08-27:
I dont think HPER_010 is redundant as it provides a console for the OS and standardizing on the UART avoids the OS from having to carry and qualify various possible custom UART drivers. Secure boot does matter because to the OS do seal secrets such as disk encryption keys to their measurements.

I suggest we expand RVA_070 with following rules:
1. All implemented triggers MUST 
      - Support privilege mode filtering (VS, VU, S, U).
      - Meet the requirements for configuring action=0.
      - Support matching all legal addresses using instruction and load/store address triggers.
      - Support filtering using mhselect values of 0, 1, and 2.
      - Support filtering using sselect values of 0, 1, and 2.
      - Support filtering using all supported values of scontext, hcontext, ASID, and VMID.

ved-rivos, 2025-08-31:
> he prior requirement was 0 and either 1 and/or 2 (i.e. 0+1 or 0+2 or 0+1+2).

Here we had agreement on 0+2. We can include 0+1 as an option.

ved-rivos, 2025-08-31:
Please see PR #92

andreiw, 2025-09-08:
Could we also clarify if the mandated trigger counts are per-mode or a grand total? I get that the we're talking about S-mode software observable counts, but this still leaves S vs U vs VS vs VU...

closed 2025-09-10 by jones-drew