riscv-non-isa/riscv-server-platform #93: minor: dead link in about section

jchzhou, 2025-09-05:
The link in the About section of this repository, https://jira.riscv.org/browse/RVC-6, is no longer accessible.
Please update it with the new link https://riscv.atlassian.net/browse/RVS-1781

jones-drew, 2025-09-05:
@jchzhou This specification doesn't have an "About" section nor do I find the URL you reference anywhere in it. Please send a PR with your suggested change.

ved-rivos, 2025-09-05:
@jones-drew - its the "About" section of the GitHub repo. It has the old jira link that is now replaced by the atlassian link.

[image]